itsjoshpark/charge-limiter #1: Path to smcutil hard coded

alexwhittemore, 2020-05-28:
I saw your post over on the Al Dente issue and figured I'd give this a try, since I didn't understand exactly how you'd made it from your description. I tried to run it from my downloads folder and got this: 

[image]

godly-devotion, 2020-05-28:
Sorry about that, relative paths are a tricky thing. I've made some updates and released a new version (same version number though). Try that version and let me know if that works.

alexwhittemore, 2020-05-28:
Sure looks like it works - I like it! 

I know I did a little funnybusiness on the Alfred workflow - I actually wrapped the terminal command call in applescript so that osascript would be responsible for privilege elevation - that way, it prompts you either for a password or for touch ID if you have it. I wonder if there's an easy tweak to avoid typing a password in this case if you don't have to? Anyway, totally keeping this around, thanks!

Random aside - it looks like, if you set the max limit to a value lower than your current state of charge, charging behavior won't change until you re-plug power. Like the PMIC only checks the SMC value on power-attached events.

godly-devotion, 2020-05-28:
Awesome!

As you mentioned, I did notice that I had to drain my battery a few percentages below the set max limit in order to verify that it would stop charging correctly.

closed 2020-05-28 by godly-devotion