itsjoshpark/charge-limiter #4: Is it better to use this or Catalina's health management feature?

sn0wyfall, 2020-06-09:
Wondering if it's better to use this or to just use Catalina's health management feature. Any ideas?

sn0wyfall, 2020-06-09:
Also, what happens if you keep both of them on at the same time? too scared to try that out

Knightfall77, 2020-06-09:
> Wondering if it's better to use this or to just use Catalina's health management feature. Any ideas?

Personally I would use the built in feature rather than this. This is for older OS's or for folks who don't want Catalina.

sn0wyfall, 2020-06-09:
> > Wondering if it's better to use this or to just use Catalina's health management feature. Any ideas?
> 
> Personally I would use the built in feature rather than this. This is for older OS's or for folks who don't want Catalina.

That's a fair point. But I'm wondering if Apple favors convenience more. Of course Apple would never release a feature where the charge level is perpetually 50%, but wouldn't that be better for the battery than leaving it at 100% for 96% of the day?

godly-devotion, 2020-06-09:
I think it really depends on your work habits and how much convenience you'd like.

If you are someone who takes your laptop to many different places, the native health management feature of Catalina can be definitely easier to use because there are no options and it keeps it close to full charge.

As @sn0wyfall mentioned, Apple's solution is something for the most common scenario of its users. Everybody has different use cases. After all just because keeping the battery charged less is good doesn't mean it will work for those who need all the charge they can get.

I personally use my laptop connected to power most of the time so I leave the battery charged to only 50-70%, and on the few days I take it with me I make sure to allow charging to 100%.

So in the end, it really depends on what your work pattern is and how much convenience you will trade off for longer battery life.

sn0wyfall, 2020-06-09:
@godly-devotion Awesome, thanks for the reply! For the time being I'm going to be running my laptop off AC power. So would it be _better_ for now to switch to limiting my charge (i.e. better than Apple's health management feature)? I've noticed it increases battery cycles, albeit slowly.

godly-devotion, 2020-06-10:
Probably better to limit to 50% as I’m doing then. That way you won’t be increasing your charge cycle count even albeit slowly.

sn0wyfall, 2020-06-10:
Awesome, that's what I figured too. Thanks a lot!

closed 2020-06-10 by sn0wyfall