itsjoshpark/charge-limiter #5: What happens if you set a charge limit using this and enable Catalina's health management feature?

sn0wyfall, 2020-06-12:
Say I limited my charge to 50%, and used Catalina's health management feature. What would happen? Would it still charge/discharge? Or would that behavior stop?

godly-devotion, 2020-06-12:
I am not running Catalina so I don't know. If I were to take a guess, the health management feature will probably not kick in if you limit your charge to 50% because it is no where near full charge since that feature is mainly there to not fully charge the battery until you need it.

sn0wyfall, 2020-06-12:
thanks mate, mind if I wait for someone else to chime in to answer this question?

sn0wyfall, 2020-06-12:
I went and enabled it while on the management feature. I set it to 80%. I'll notify y'all what happens when the battery next start going down. Currently at 100%. Wish me luck

sn0wyfall, 2020-06-13:
Okay, so the battery level went down to 92% and is staying there. Usually it cycles between 100% and 89%. I'm going to give it another day and see what happens—will it decrease even more, or will it just stay there?

sn0wyfall, 2020-06-16:
Okay, seems like it stops. Maybe the battery health management feature only works if you're actually charging to 100%.

closed 2020-06-16 by sn0wyfall