itsjoshpark/charge-limiter #8: error when setting value: any hardware limitations?

sdedalus1, 2020-07-07:
I've tried the script on a 2008 aluminum unibody Macbook running Sierra 10.12.6, but after entering a max charge value I get "There was an error setting the charge limit. Make sure you have administrator privileges." This is with the correct admin password.

This model doesn't officially support Sierra, but I have installed it with DOSDude's macOS Sierra Patcher Tool for Unsupported Macs. Just wondering if there might be a software workaround or it could be a hardware limitation.

godly-devotion, 2020-07-08:
Its hard to say. Technically the `smcutil` thats included in the app (which does the setting of the value) supports up to 10.12. There is a version of the utility that supports older versions of macOS but I haven't yet figured out how to compile it.

I'm curious as to why its failing. Please try the following:
1. Open Terminal window.
2. Copy the following command: `sudo /Applications/Charge\ Limiter.app/Contents/Resources/smcutil -w BCLM 64`. Modify the path of the Charge Limiter app if its not under the Applications folder.
3. Press Enter. This will set the max charge limit to 100. Let me know what errors show up if any.

godly-devotion, 2020-07-26:
Closed as no response.

closed 2020-07-26 by godly-devotion

sdedalus1, 2021-02-12:
Sorry, for some reason I missed the notification that you had responded! Here is the only thing Terminal spits out (and I have confirmed the path to smcutil is correct):
`Error: write error`

godly-devotion, 2021-02-12:
It’s quite possible it’s simply not supported on that older model. Unfortunately I didn’t create the smc utility which does the heavy lifting so I don’t think I’ll be able to add support to your MacBook.