itsjoshpark/charge-limiter #9: Resets after restart

sn0wyfall, 2020-07-12:
Hi,

I've noticed that my laptop starts charging again after a reset, without my having to reset the SMC. Does this happen for anyone else?

godly-devotion, 2020-07-12:
That is possible after a reboot, although not necessarily after every reboot. The MacBook will sometimes decide to reset parts or all of the SMC. How it decides that I don't know.

A potential long term solution to this would be to install a launch daemon or helper that will reset the desired charging level after a reboot like how AlDente does. A simpler way would be to remember the user selected charge level and set the value at boot by having the user add it to one of their Login Items.

sn0wyfall, 2020-07-13:
Nice! Didn't know that. So that's why AlDente requires a helper tool. Good to know, thanks.

For now I'll start it on login.

Donothir, 2020-08-06:
Interesting, is it in your plans to add a deamon or an helper?

Having a simple, one step, to control the charge level (or better, a pref pane) would be great. I looked at the AlDente software, but it's not retro compatible (10.13), so your application seems better suited to my needs.

godly-devotion, 2020-08-06:
To be honest I probably wont considering the amount of time I currently have. This script app only took a few hours to create but a full blown app or pref panel will require more time and investment. The project is open source so anyone can feel free to look at the code and take some ideas on improving it or creating their own thing.

Donothir, 2020-08-06:
You've done enough, I was just checking if you planned to do more, no obligations!
Thanks!

closed 2021-11-03 by godly-devotion

godly-devotion, 2021-11-03:
Persistency on restart was added to the latest version. Please note that in order for this to work properly, the app should be first copied to a place such as the Applications folder so that it can reapply the charge limit on boot.

commit 3cd51506 (2021-11-03): Replace `smcutil` with `bclm` for better compatibility

Fixes #30. Closes #9.