itsjoshpark/charge-limiter #10: Charge indicator LED on Magsafe connector

fnarfbargle, 2020-10-26:
While this software works brilliantly to limit the final charge on the battery, on my MacbookPro 11,1 the charge led on the magsafe connector always remains orange.

After a bit of a poke around on my machine the BFCL key sets the level at which the SMC turns the led green.
So to set the battery limit at 75%, BCLM=4b. If I set BFCL=4a then the battery charge led works correctly.

Hope this helps.

godly-devotion, 2020-10-26:
Hmm, do you know what values the `BFCL` is taking? Is it a hex value?

fnarfbargle, 2020-10-26:
It is on my machine. Takes the same format as BCLM. By default it was 5f. I found as long as it was at least the BCLM value -1 it did the job.
I simply set it with smcutil -w BFCL 4a
This is an excerpt reading the SMC in Linux (the integers at the end are simply the index into the list, not values) :
```
BCLM  [ui8 ]  4b 80
BFCL  [ui8 ]  4a 83
```

godly-devotion, 2020-10-26:
Interesting. Seeing how the `BFCL` value is defaulted to `5f` (95%), I wonder if it should be safe to set it to `BCLM`-5. At least then when the user resets the `BCLM` back to `100` we will revert the `BFCL` to `95`. I will test this out on my machine and if it looks good I will add this feature to the app. Thanks for your help!

closed 2020-10-26 by godly-devotion

godly-devotion, 2020-10-30:
@fnarfbargle It seems that the `BFCL` value doesn't exist for some combo of MacBook & macOS. Could you tell me what version of macOS are you running?

fnarfbargle, 2020-10-31:
Product Name: MacBookPro11,1
OSX : 10.15.7
Boot ROM Version:	162.0.0.0.0
SMC Version (system):	2.16f68

I assume I'm running the latest firmware and EFI. I'd have thought if it supported BCLM it'd support BFCL, but perhaps the hardware or SMC firmware doesn't support it on some models.

godly-devotion, 2020-10-31:
Hmm I’m not seeing a pattern as to which ones have BFCL available. Thank you anyway.

commit e0969af2 (2020-10-26): Display correct MagSafe LED indicator color (version 1.3)
Closes #10