itsjoshpark/charge-limiter #11: version 1.3 can't change level on MacBook Pro 15" 2012 (older works 1.21)

iPlasm, 2020-10-30:
in version 1.3, after entering password this message comes up:
"There was a problem setting the charge limit. Please make sure you have administrator privileges."

but in older version works v1.21, no such a messages and works fine. 

MacBook 15" 2012 MacOS Catalina 10.15.7

Thanks

godly-devotion, 2020-10-30:
Could you please try running this command in Terminal and copy what result you get here?

`"<path-to-charge-limiter.app>/Contents/Resources/smcutil" -r BFCL`

manasv, 2020-10-30:
I'm having the same behaviour, the command you gave returned `no data` for me.

godly-devotion, 2020-10-30:
@manasv Could I get your MacBook model and OS version?

manasv, 2020-10-30:
I'm on MBP 16" Base Model and right now on beta (Big Sur 11.0.1 Beta)

(The past version works as @iPlasm said)

godly-devotion, 2020-10-30:
Could you please try this build and see if this fixes the issue for you?

[Charge Limiter (for issue 11).zip](https://github.com/godly-devotion/charge-limiter/files/5468076/Charge.Limiter.for.issue.11.zip)

iPlasm, 2020-10-30:
> Could you please try this build and see if this fixes the issue for you?
> 
> [Charge Limiter (for issue 11).zip](https://github.com/godly-devotion/charge-limiter/files/5468076/Charge.Limiter.for.issue.11.zip)

Yes that works, the message said successful. (but I need more time to see the results)
for optional, I like that you add an icon for it, it's easier to know and nice to have. 
I love this icon, unless if copywriting is issue (not sure) https://icon-library.com/icon/android-battery-icon-18.html

For v1.21, I tested on MacBook 12" 2017 and MacBook 13" 2012 (both Macs 10.15.7), they were both worked great at the percentage battery level I want.
Today I was tried to active on my MacBook 15" 2012 too, so it was chance that I faced this update notify.
 Thank you

godly-devotion, 2020-10-31:
Thanks for the info! I've released version 1.31 with the fixes included.

closed 2020-10-31 by godly-devotion

iPlasm, 2020-10-31:
> Thanks for the info! I've released version 1.31 with the fixes included.

Yes, I confirm it is now working on at the level battery I want on MacBook 15" 2012 MacOS 10.15.7.

Thank you

commit 213a258e (2020-10-31): Don't crash if BFCL value doesn't exist

Resolves #11