itsjoshpark/charge-limiter #17: Thunderbolt / USBC port only receives power, Not data.

jrydinger, 2021-01-23:
Hey! I have a macbook pro 2017 with two thunderbolt ports. A problem arose a few weeks ago after installing this code where only one port can receive data such as sound interface's via firewire to thunderbolt or a midi keyboard's USB2 to USBC. The lower (closest to my chest) port works as usual, but the upper one can only receive power. I'm thinking the faulty port doesn't receive enough power. I don't know for sure this problem is caused by the code but the problem arose when installing the code.

godly-devotion, 2021-01-23:
I don’t have a USB-C MacBook but you can reset the charge limit to 100 which would be the default.

godly-devotion, 2021-01-29:
Closing due to no updates and potentially unrelated to project.

closed 2021-01-29 by godly-devotion