itsjoshpark/charge-limiter #30: Getting an error when I run: Error 1004: An error occurred.

NickStrupat, 2021-11-02:
[image]

It was working for a few months before today. Hoping someone here knows how to fix this issue.

godly-devotion, 2021-11-02:
What is your MacBook and macOS version? What was the last version that worked for you?

NickStrupat, 2021-11-02:
MacBook is late 2013 (MacBookPro11,3)
macOS version is 10.15.7 (19H1217)

I believe the last version to work for me was 1.4.

The error happened after I updated today to 1.4.1, but I'm not sure exactly if that was the cause.

I installed 1.4 again and I still get the error.

jenyunhuang, 2021-11-02:
I have same issues
macbook pro 2015 15" catalina 10.15.7

NickStrupat, 2021-11-02:
I wonder if it's a `zsh` issue. When I run `sudo /Applications/Charge\ Limiter.app/Contents/Resources/smcutil -r BCLM` I get `illegal hardware instruction` followed by the command text.

godly-devotion, 2021-11-02:
Could you please try this test build to see if it works? @MikeMaxNow If you don't mind, could you test this build to see if it works for you as well?

[Charge Limiter.zip](https://github.com/godly-devotion/charge-limiter/files/7464612/Charge.Limiter.zip)

jenyunhuang, 2021-11-03:
The new file work for me

NickStrupat, 2021-11-03:
Works for me too.

I did some thinking and realized it's probably that the embedded `smcutil` was compiled on a newer Mac with some instruction set extensions that aren't on my 2013 CPU. When my machine tries to run the instructions it doesn't know, it yields an `illegal hardware instruction` error.

MikeMaxNow, 2021-11-03:
It is working for me!

godly-devotion, 2021-11-03:
Awesome, thanks!

commit 3cd51506 (2021-11-03): Replace `smcutil` with `bclm` for better compatibility

Fixes #30. Closes #9.
closed 2021-11-03 by godly-devotion