itsjoshpark/charge-limiter #31: Error - Symbol Not Found

quickdry, 2021-11-05:
Error on line 22: Error: dyld: Symbol not found: _$s17StringLiteralTypes013ExpressibleByaB0PTl
  Referenced from: /Applications/Charge Limiter.app/Contents/Resources/bclm
  Expected in: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/swift/macosx/libswiftCore.dylib
 in /Applications/Charge Limiter.app/Contents/Resources/bclm (1005)

Error on MacOS El Capitan , MBP retina mid 2014 i7 3.1GHz

godly-devotion, 2021-11-05:
macOS El Capitan provides an outdated version of Swift that isn't compatible with the utility I am using (`bclm`). One solution to this issue seems to be to install [Swift 5 Runtime Support for Command Line Tools](https://support.apple.com/kb/dl1998?locale=en_US).

See zackelia/bclm#6 for details.

godly-devotion, 2021-11-17:
Closing due to no updates.

closed 2021-11-17 by godly-devotion

arcjive, 2022-03-11:
Installing this Swift5 package fixed my similar issue with v1.5 on macOS Sierra. v1.4.1 and earlier worked fine.
You should link this package on the main page and note as a requirement for earlier OS's...?
Thanks for the great little utility, very useful.

godly-devotion, 2022-03-11:
Good idea.