itsjoshpark/charge-limiter #33: There was a problem setting the charge limit

florin-hcl, 2021-11-30:
So this app worked with no problems until now. Suddenly, it says "There was a problem setting the charge limit. Please make sure you have administrator privileges." and this is my own computer and I'm the admin... I own it... no problems... until now. For some reason I can't change the charge limit in v1.5.

godly-devotion, 2021-11-30:
Is this a new problem after updating to v1.5 or were you using this version before without issue? What is your MacBook and macOS version?

florin-hcl, 2021-12-01:
This is a new problem after the latest update. I'm using macOS Monterey 12.0.1 on a MacBook Pro 13'' i5 2016

godly-devotion, 2021-12-01:
If you open Terminal app and paste this command, what's the output? (It should set the charge limit to 80%)

sudo /Applications/Charge\ Limiter.app/Contents/Resources/bclm write 80

florin-hcl, 2021-12-01:
After I started getting that error, the limit was removed and the battery charged to 100%.

I reinstalled the app, I typed that command in Terminal and I got the usual macOS popup about unverified developers, I went to Security and opened Charge Limiter from there. I got the same error as before.

Then, I typed the same thing in Terminal again and I didn't get any kind of message or output (except sudo asking for credentials). I let my battery discharge down to 77%, I plugged the laptop back in and now charging stopped at 83%. When I click on the battery icon in the top bar it says "Battery Is Not Charging".

The command line worked fine! It feels like it's something related to the GUI.

triumphantomato, 2022-07-14:
i had the same error - it looks like the gui prompts for a value between 40-100, but the command line prompts for a value between 50-100.

i entered the gui again and tried 50 and got a "success!" message

this is on Catalina

jcherrera, 2023-01-24:
> sudo /Applications/Charge\ Limiter.app/Contents/Resources/bclm write 80

keyNotFound(code: "BCLM")

on macos 13.2

charly32, 2023-12-05:
macbook air 11 2013 big sur 11.7.10

with trick "sudo /Applications/Charge\ Limiter.app/Contents/Resources/bclm write 80"

Success!

If your MacBook currently has more than 80% charge, leave the charger unplugged until the battery level is a little less than that. Afterwards, you can leave the MacBook connected to the charger and the battery will not charge over 80%.

thanks

amlboeton, 2024-05-17:
sudo /Applications/Charge\ Limiter.app/Contents/Resources/bclm write 80

keyNotFound(code: "BCLM")

on Ventura 13.6.7