itsjoshpark/charge-limiter #34: keyNotFound(code: "BCLM")%

AdinAck, 2021-12-24:
When running the app, is says the current charge is "keyNotFound(code: "BCLM")%".

It also always gives the warning that my battery is above the set charge limit even if it isn't.

isaacnorwich, 2022-01-21:
I am also having this error. M1 Macbook Pro with Apple M1 Pro chip on macOS Monterey. Same things happens to me when i type the same command `sudo /Applications/Charge\ Limiter.app/Contents/Resources/bclm write 80` in the terminal.

AdinAck, 2022-01-21:
I have learned that the M1 does not have an SMC, and thus there is no key to write to. Unfortunately there is no option for us :(

I did have an idea to overwrite the ML model that determines when to hold at 80% with one that always outputs True, and I have created the file, but cannot figure out how to overwrite system files.

isaacnorwich, 2022-01-21:
How does AlDente (https://github.com/davidwernhart/AlDente) work then with M1 Macs? Thanks for the info though!

AdinAck, 2022-01-21:
It just disconnects the charger until the desired battery level is reached. The unfortunate thing about this is that clamshell doesn't work with it as clamshell is only enabled when connected to power, that is why I resorted to forcing the ML model to handle the SOC decrease.

Danby108, 2022-07-04:
> I have learned that the M1 does not have an SMC, and thus there is no key to write to. Unfortunately there is no option for us :(

All the functions the SMC handled are now embedded in the M1 processor. That must include recharging the battery.
So the lack of an SMC does NOT mean there's no way to limit the battery charging level.

It does explain why the two utilities I've tried, which reference BCLM, don't do what they were intended to do on my M1 Mac.

Clearly a new version of charge-limiter is needed, which will:
First determine if it's running on an older mac with an SMC, or a newer M1 Mac.
If it's running on an older mac, then it should apply the existing code.
If it's running on a newer M1 Mac, then it should apply a new approach, which ought to be possible.

If godly-devotion is willing to attempt this new version, I'd be willing to sponsor him/her/them with a $20 contribution.

Danby108, 2022-07-04:
Update: Limiting the battery charge is DEFINITELY possible on the M1 machines, since the AlDente app is successfully doing it. However, the charging indicator light is incorrectly showing that it's charging when it's not.

Laurensvl, 2025-08-04:
I have the same problem with a MacBook Air M3, so I guess that renders this app broken on more modern Macs?