itsjoshpark/charge-limiter #37: ChargeLimiter stops working on macOS Sierra after installing "Device Support Update"...

arcjive, 2022-03-25:
Previously was working. Recently performed a Software Update, and installed the Apple "Device Support Update" (I believe it was posted around Sept 2021, I just hadn't checked in a while).

Since then I receive this error upon launch;
The operation couldn’t be completed. (OSStatus error -192.)
(null) (-192)

Previously I received this same error when attempting upgrade Charge Limiter from 1.4.1 to 1.5.  Installing the Swift5 runtime package fixed it.
Unfortunately now, even after re-installing the Swift5 package, NONE of the versions I used to use work either!  I've tried 1.3.1, 1.4.1, and 1.5.  All with the same error.

Is there anything we can check to see what's causing this?

Thank you.

Zibroff, 2022-08-12:
Hi!
Is your issue resolved?

arcjive, 2022-08-12:
Yes and no.  I did a clean re-install of MacOS Sierra, **didn't install the Apple "Device Support Update",** and stuck to v1.4.1, and it works fine. If I upgrade to 1.5,  or install the Device Support Update, it will break again.

Zibroff, 2022-08-12:
Thanks for the flash answer!

May be sufficiently roll-back iTunes version instead re-install MacOS overall?
I've the same issue after update iTunes with Device Support Update.
And I'll try roll-back to previous version iTunes.
Inform you later.

P.S.
Sorry for my English. I'm from Ukraine.

Zibroff, 2022-08-15:
@arcjive
I've decide to use this CLI tool:
https://github.com/zackelia/bclm

arcjive, 2022-08-15:
Thank you, that looks nice and simple.