itsjoshpark/charge-limiter #42: cannot set charge limit to 100

foxnoobkite, 2022-09-28:
when i tried to make it 100 it keep saying "There was a problem setting the charge limit. Please make sure you have administrator privileges." how do i fix it

geobio, 2022-11-16:
do you get an error if setting it to some other percentage like 95?

geobio, 2022-11-16:
I'm on an 2021 m1 macbook pro, and get the error with 100, but no error with 99 or below.

If you're on an m1 macbook, it may actually be impossible to use this charge-limiter, see: 
https://github.com/godly-devotion/charge-limiter/issues/34

foxnoobkite, 2022-11-16:
i'm on m2 mac, it is very annoying every time i have to go to battery icon and set "charge to full tommarow", is there a way to disable this app?