itsjoshpark/charge-limiter #44: does not limit charging on MacBook Pro retina mid 2015

jb-gha, 2022-11-16:
just installed this after replacing the battery.
Set max load to 80%. but the battery still is loaded over 80%. did a restart too.
the led on the charger is green, but in menu icon I see it is charging over 80%.
Any hints?

jb-gha, 2022-11-16:
hmm. it stopped charging at 84% at a setting of 80%

porteusconf, 2023-01-16:
It is a known-issue that charging does not actually stop until a few % more than the limit. This app uses `bclm` under the hood, and the [Readme.md for bclm ](https://github.com/zackelia/bclm/blob/master/README.md) says: 
> "When writing values, macOS charges slightly beyond the set value (~3%). In order to display 80% when fully charged, it is recommended to set the BCLM value to 77%."

So, it appears this discrepancy is not due to any software bug, and the bclm author apparently does not want to build-in any tweak to set limit a few % more than what you ask for. And I think it is a good idea, for many reasons to just set bclm to the value you specify rather than some slightly lower value. That is, it should be up to you rather than the program to choose the slightly lower limit to set. So, there is nothing to fix, other than perhaps just adding a  suggestion somewhere like:
> "Set the limit a few % less than you actually want, like 76% or 77%  if you actually want 80%. YMMV depending on battery condition. 

FWIW, on my 2012 mabook-air that says "Service Battery" (it only lasts 10-20 minutes), it does not stop charging until 93% when bclm is set to 89% (below), perhaps since battery is in poor condition.
``` shell
bash-5.0$ bclm read
89

### still charging with battery at 90%, 91% and 92%
bash-5.0$ pmset -g batt
Now drawing from 'AC Power'
 -InternalBattery-0 (id=5767267)        90%; charging; 0:32 remaining present: true

### "pmset -g batt" finally says "AC attached; not charging" at 93%
bash-5.0$ pmset -g batt
Now drawing from 'AC Power'
 -InternalBattery-0 (id=5767267)        93%; AC attached; not charging present: true
```

jb-gha, 2023-01-17:
Thank you for the detailed explanation.
For me, this behavior ist perfectly fine.