itsjoshpark/charge-limiter #52: The battery does not charge properly

Ayespe, 2023-03-09:
app is v1.5.1
The system version is 10.14.6
At the beginning of using the software, it is normal to set the charging power. After using it for a while, today I unplug the charging cable and use it for a while. After charging again, it will display "the battery is not charging". Now the mac has only 25% of the power, but it still cannot Continue to charge normally, but it can only be charged up to 25%, and the reset does not work, what should I do?
Using the code sudo /Applications/Charge\ Limiter.app/Contents/Resources/bclm write 80 will prompt me to enter the password, and there is no abnormal output after pressing Enter, but the problem of not being able to charge still exists
[https://i.imgur.com/ONTWD8j.jpg](url)

Ayespe, 2023-03-09:
When this charging problem occurs, I set it again and it will display "There was a problem setting the charge limit. Please make sure you have administrator privileges." This prompt, after several operations, will now return that the setting is successful, but it is actually invalid

curtnorris, 2023-03-10:
app is v1.5.1
The system version is 10.13.6

Similar issue. Was working fine for many months with CL v1.31. At some point, battery percentage started dropping well below the set threshold of 70%. Installed Swift Runtime and ran version 1.5.1 and get the same error, "There was a problem setting the charge limit. Please make sure you have administrator privileges."

Ayespe, 2023-03-10:
这是来自QQ邮箱的假期自动回复邮件。您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

curtnorris, 2023-03-11:
UPDATE: I resolved the issue of the battery not charging at all by resetting the SMC. Went back to CL v1.31 to set the charge limit percentage and the charge limiting is working again.

I don't know the cause or the reason newer versions of the app aren't working on this MacBook but it's doing what I need it to do so I'll count that as a win.